kilpatrickaudio/CARBON #7: General wishlist

immerfroehlich, 2016-07-31:
Hi Andrew,

after having the Carbon the very first weeks and playing with it the
first few hours, I must say that I really like it.
Especially the
- Step sequencing/recording mode (expandend upon the sequencer of the Phenol)
- I really like the workflow of the Step Editing.
- Grafical view
- One button one function (or one concept).

What I like to do with the carbon is to record/step edit a
rythm track and a base track and a drum track and then play
along with it and maybe record that playalong too.

And for this workflow I have a few wishes that I like you to
consider for the next versions. Some things may be implemented
already but it may be that I don't understand the concept.
Here are my points:

- What is a scene? Just a selection of a playback area of the
  64 steps? If that is the case I think it would be better
  if a scene could be a different "recording". For Example
  Scene 1 contains the notes A-B-C (each on a single step) and
  Scene 2 contains the notes D-E-F. In that case it would be nice
  to give names to the scenes. For example "Intro (Scene1)",
  Chorus (Scene 2), Outro (Scene 3), etc.

- Only 64 steps per track (if I got it right) - that is very limiting.
  Unlimited steps per scene would be nice.
  Display: If you go beyond the last visible step the next line will
  be displayed on the bottom and the top line disappears (like a stack or something).
  There could be a number written on each step, so you can see on which
  part of the scene you are at the moment.

Song Mode:
- I don't understand what the number right after the scene is.
  The manual says "LENGTH - adjust the playback length for the scene".
  But what exactly is the length of the playback? At the moment
  when I set it to 1 it plays back 4 steps of the selected scene.
  I think it would be nice if the number would be representing
  either the number of times the scene will be played or the number of
  steps (of the scene) that will be played.

Recording / Editing Mode
- It would be nice to select the step when in "Step Recording" mode.
  This way you could quickly correct a wrong entered note.
- Would be nice to select how long a entered note will be played.
  It would be nice if musical terms could be used (4th, 8th, 16th, etc).
- I'd like to select how long a step is (4th, 8th, 16th, etc - maybe
  implemented already, but I did't find it).
- And it would be nice to select how long the next entered note
  will be played. For example if you change the step length to 8th
  and you select the entered notes to be played 8th long than all
  entered notes will be one step long until I change the length of
  the next entered note.
- It would be nice to insert or delete steps when in editing mode.
- The realtime recording should't stop, when all steps are played.
  It would be nice to correct the recorded notes. Either on a new
  track or the recorded notes will be overwritten.

Playback Mode
- When in Playback Mode it would be nice to transpose the notes
  when pressing a key on the midi controller (keyboard). Just
  like it is on the Phenol.
- During playback I like to play my MIDI- Controller (keyboard)
  and hear those notes on the connected synthesizer. But all
  input is muted on playback.

General
- The way the menus disappears after a few seconds is not very
  statisfying. For example when playing the arpeggiator in live
  mode I'd like to play with the settings but when I listen to
  it the menu disappears after a few seconds. An exit button
  would be nice. So I have the power to exit the menu when
  I want to. The same behaviour can be seen on a few modes for
  example when in playback mode and edit mode, the edit mode will
  disappear after a few seconds.

Bugs:
- After the last step is played (in normal playback mode),
  you can hear a short delay before the first step is played back.

OK, that's all for now. I like to hear your opinion about it.

Kind regards,
Andreas

kilpatrickaudio, 2016-08-02:
Hi Andreas,

Thank you for your comments and questions. I'll try to answer them below, and will subsequently update the manual to reflect your questions as an aid to others.

Scenes: The scenes store all parameters on tracks such as the start and length, pattern selection, and many other things. A full list of parameters will be published in the manual on the next update. The reason why it is not possible to record different notes is that after much testing and experimenting it became clear that it wasn't actually very much fun. The original design had new notes per scene, and indeed the code still supports compiling a version that supports it, but it's not very fun in real life to use which is why the current behaviour was chosen.

64 steps per track: Yes, CARBON is a step sequencer. The visual organization as well as the limitation is designed on purpose. If this is a really big problem for enough people we will consider releasing an alternate firmware that uses different notes per scene as a way to expand it, but as mentioned above, this turned out to be not much fun. Remember CARBON is not intended to be a DAW... it's a fun and fast step sequencer that helps get your creative juices flowing. :)

Song Mode: The length parameter is the number of beats. Since CARBON doesn't know about time signature the beat is the base unit for time.

Recording / Editing:

Selecting the step: The step is selected with the SELECT control when in step edit mode.

Note length: This can be adjusted using the GATE TIME control. There is currently no way to adjust how long the steps will be when they are recorded, other than selecting the step size for the track.

Step length display: This is shown in the step edit mode if you turn the GATE TIME control. By default it shows the note velocity.

Entered note length: We will look into this.

Insert / delete notes: We will look into this.

Recording shouldn't stop automatically: This is an interesting point and we can look into implementing this.

Playback mode: In both cases please investigate the LIVE and keyboard transpose modes. They do what you're asking.

General: We can look into leaving menus on indefinitely, but were dismissing them with a timer so that the screen returns to a known state automatically.

Bugs: Please explain your track setup and approx. how many notes are in use when the playback comes around. We spent a lot of time verifying timing and believe it is correct. Since it's a step sequencer with quantized notes, many events may get sent out at once. Because MIDI is slow, this can cause a perceivable delay on higher numbered tracks. We are working on a function to allow drum tracks to be sent out first, to minimize the perception of any latency. This will be included in the next release.

Thanks very much for taking the time to comment on CARBON! We've taken note of your ideas for the next round of updates planned to be started shortly.

Andrew

negomozambique, 2016-08-15:
Hey Andrew, 
First of all, I'd like that I am amazed about the Carbon Sequencer. I understood the concept and I feel like placing some sugestions:

**Arpeggiator** 
A better way to edit  or to get more options, rhythmically speaking.

**Knobs/Encoders:** 
Real time recording from them, not just editing in step mode. Transpose variations, Gate time variations, switching arpeggios... 

**Patterns** 
More Patterns options for sure, they are sooo cool!!! Another great thing would be if the patterns have the ability of trigging other parameters, not exclusively mutes. You could apply a Pattern triggering pre-defined parameters over the played notes. A Random Patterns generator would be nice too ( press PATTERN+SHIFT+DIR and you got a Random Pattern, lol)

**Sequencer** 
Well, I love the idea of recording a phrase and done: the sequencer stops recording. But I agree with Immer, sometimes you don't wanna press "Live"+"record again" etc... we should have another sequencing mode, more choices... It would be nice keeping the recording mode runing while I am listening and then, play other events... Even I have found the two modes (drum and voice) very good, I'd like this "wasting time" option. 

**Internal Metronome**
 Option to regulate the volume. 

**Scenes** 
So, you can copy scenes and you can not erase one channel and record another phrase, right? If I do this procedure, Carbon will repeat the same new phrase in all scenes, right?
You can edit it, you can pattern it, you can even mute this channel, but you can not write a new phrase. So, I would like more independence copying scenes and tracks. 12 Scenes would be nice too but, I agree with you: it's hard to make anything bigger than "six paths for everything" in Carbon, just because it's not the same fast flow. 

But, try to understand our point here, this Sequencer is fantastic for what was build for:  to make music by itself. That's why I love it. But still, I work with it (music) and I need both options. I want to write new notes over my copied scenes and then, I want to use again all the Carbon stuff over it. Without changing my events in the first scene! Why not? Sometimes the machine has the answer that you're looking for, sometimes you have the "notes-melody-ideas" just on your finger tips... Why Carbon should limit that? The idea of "we tested and wasn't funny enough" does not sound a good argument for me. Both ways are funny. And since Carbon, I can't think about writing something without trying all patterns over it. lol. 

Freedom increases creativity. I don't know if is possible... But I would go for 12 scenes (with 6 tracks), ability to copy tracks, ability to record new notes. 

Another point: I noticed when I am jamming with the scenes, sometimes there is a delay (perhaps my hands, perhaps Carbon) which doesn't happens in the song mode... For example, I have two different scenes, different direction in the same channel and boom, I loose the clock! So, would be nice more modes of playing the scenes live, to make sure that they never loose the clock (I use to make live gigs and I really want to bring Carbon to this... In this way is a little bit hard to jam with the scenes in a live act, for example. It works for sure, but could be tighter).

**Non-Registered Parameter Number (NRPN)** Why not?

Peace!
Nego Mozambique

CheckMate808, 2016-09-18:
Dear Andrew,

First of all I'm really happy to own a Carbon sequencer and it is really a great tool and lots of fun to use. Especially with the CV/Gate outputs in combination with my modular.
Referring to Andreas comment I'd also like to exit menus on my own wish. 
The time in "step edit" is way too short and it gets quite annoying when trying to edit notes and thinking a bit too long.
I'd be really happy if you could implement that i.e. "Pressing shift+step edit" exits step edit mode.
I'd also like an easier way to quickly correct a wrong entered note and inserting or deleting steps when in editing mode.
+1 also on realtime recording not stopping, when all steps are played. 
If possible an Overdub function would be great similar to the MPC sequencer.
I sold my MPC because editing notes and events was really slow which is way better with Carbon.
Best,
Enrico

Gnapp, 2016-09-18:
+1 on making Step Edit mode permanent until I chose to leave edit mode.
+1 on making Realtime recording loop (with overdubbing) until I chose to leave recording mode.

These two features would remove the only annoyances I find in the Carbon. The rest is perfect.

BR

Conny

kilpatrickaudio, 2016-09-18:
Thank you for your comments. Please only post original points as it takes a long time to read through all posts only to find that we already had some items on our todo list. :) We take each comment with equal weighting based on its merits.

closed 2016-09-18 by kilpatrickaudio