kilpatrickaudio/CARBON #8: Carbon is amazing: Sugestions here!

negomozambique, 2016-08-15:
Hey Andrew, 
First of all, I'd like saying that I am amazed about the Carbon Sequencer. I understood the concept and I feel like placing some sugestions:

**Arpeggiator** 
A better way to edit or to get more options, rhythmically speaking.

**Knobs/Encoders:** 
Real time recording from them, not just editing in step mode. Transpose variations, Gate time variations, switching arpeggios... I wanna record everythong, in realt time.

**Patterns** 
More Patterns options for sure, they are sooo cool!!! Another great thing would be if the patterns have the ability of trigging other parameters, not exclusively mutes. You could apply a Pattern triggering pre-defined parameters over the played notes. A Random Patterns generator would be nice too ( press PATTERN+SHIFT+DIR and you got a Random Pattern, lol)

**Sequencer** 
Well, I love the idea of recording a phrase and done: the sequencer stops recording. But I agree with Immer, sometimes you don't wanna press "Live"+"record again" etc... we should have another sequencing mode, more choices... It would be nice keeping the recording mode runing while I am listening and then, play other events... Even I have found the two modes (drum and voice) very good, I'd like this "wasting time" option.

**Internal Metronome**
Option to regulate the volume.

**Scenes** 
So, you can copy scenes and you can not erase one channel and record another phrase, right? If I do this procedure, Carbon will repeat the same new phrase in all scenes, right?
You can edit it, you can pattern it, you can even mute this channel, but you can not write a new phrase. So, I would like more independence copying scenes and tracks. 12 Scenes would be nice too but, I agree with you: it's hard to make anything bigger than "six paths for everything" in Carbon, just because it's not the same fast flow.

But, try to understand our point here, this Sequencer is fantastic for what was build for: to make music by itself. That's why I love it. But still, I work with it (music) and I need both options. I want to write new notes over my copied scenes and then, I want to use again all the Carbon stuff over it. Without changing my events in the first scene! Why not? Sometimes the machine has the answer that you're looking for, sometimes you have the "notes-melody-ideas" just on your finger tips... Why Carbon should limit that? The idea of "we tested and wasn't funny enough" does not sound a good argument for me. Both ways are funny. And since Carbon, I can't think about writing something without trying all patterns over it. lol.

Freedom increases creativity. I don't know if is possible... But I would go for 12 scenes (with 6 tracks), ability to copy tracks, ability to record new notes.

Another point: I noticed when I am jamming with the scenes, sometimes there is a delay (perhaps my hands, perhaps Carbon) which doesn't happens in the song mode... For example, I have two different scenes, different direction in the same channel and boom, I loose the clock! So, would be nice more modes of playing the scenes live, to make sure that they never loose the clock (I use to make live gigs and I really want to bring Carbon to this... In this way is a little bit hard to jam with the scenes in a live act, for example. It works for sure, but could be tighter).

**Non-Registered Parameter Number** (NRPN) Why not?

Peace!
Nego Mozambique

kilpatrickaudio, 2016-08-15:
Hi Nego,

Thanks for your comments. I've added your points to my todo list and will soon be transferring to using GitHub to track all the bugs and stuff. For now they are stored offline but I will respond here just so you know that I acknowledged all of them. :)

- arp - Yes we plan to have more arp programs. As for other controls I guess it depends what sorts of controls you mean. I'm not against exposing more options but so far I didn't have any.

- real-time recording of knobs - I'm not sure this fits into the current design as we really only record notes and CCs. It might be possible to have the song mode handle more parameters than currently, but I would have to hear specific things people want to do and we can see how that would work.

- patterns affecting things other than the gate - Yes we can look into this. Again what would you prefer it be able to do? Right now the patterns really just hide events on a step which is fairly core to the design of CARBON. If you have other ideas of things it could do I would be interested in considering it.

- metronome volume control - The hardware is not capable of this, but there may be a software hack to allow at least some control over the volume. The metronome was a feature added fairly late in the design so we didn't have the ability to add a complete audio system to control it.

- notes per scene - Your desire is noted. The firmware currently has a way to compile it to support notes per scene instead of notes per song. But right now that would be a global firmware setting. A lot more work will be needed to allow it to work on a per-song basis because of how the memory is loaded and allocated, but it may be possible. I decided not to do this as I already explained. I see the desire for the feature, but it was too complicated for the first release.

- clock glitch when changing scenes live - I'll have to look into this. I see no reason from having done extensive timing analysis of the running code. I believe this may be an obscure bug. Keep in mind that if you send out a lot of things on one MIDI port you might be running up against MIDI latency where changing scenes manually causes a lot of MIDI traffic. Because CARBON is quantized, many events end up getting sent out on the exact same clock tick which can cause the illusion of a glitch when it's actually just MIDI being slow. We are working on a feature that will automatically prioritize drum tracks ahead of voice tracks to try dealing with the limitations of 31250bps. :)

- NRPN - Honestly I've never used NRPN for anything... it seems cumbersome. But I will have a look and see if it's possible.

Thank you so much for your comments! We are working on getting the code up on GitHub, and are also planning another round of development within the next few months to hopefully cover most of the requests that people have made so far.

immerfroehlich, 2016-08-15:
Hi Andrew, hi Nego,

I am glad to hear that you requested the different "notes per scene" feature too. Seems I'm not the only one. This would be a really nice feature.

@Andrew
If you do prioritize things, I would vote for a menu/mode exit button first. The automatic exit function is getting in the workflow. Most of the time I am in the step edit mode. Sometimes I want to look some things up in the manual and then continue editing the notes only to recognize that I'm changing the playback length instead of changing the step. So if you change anything I hope that will be the first thing.

Arpeggiator:
Nego, that's a nice point. Andrew, do you know the arpeggiator function in the Bristol Jupiter 8 implementation? http://bristol.sourceforge.net/jupiter8.html There is a function where you can "record" the arpeggiator. But I couldn't find out how it works exactly. Must be something like if you "record" the chord it will play each different note you hold down once. But if you record a note twice it will be played twice in the arpeggio. Maybe someone knows more about it. It would be really cool if we were able to play the opening bass line of Queens "Radio Gaga". ;-) By now I couldn't find an arpeggiator that is able to do this.

So, rock on guys.

kilpatrickaudio, 2016-08-15:
Hi! I think it will be less that of priorities and more the case of "do all the easy things first" and then make a release. The menu button definitely fits into that category. The ideas that require more major reworking will obviously take a bit longer.

But I am planning to make a complete list of things that everyone is asking for and then sit down hopefully in September or so and do them all within a week or so. So please enjoy CARBON for now. :)

negomozambique, 2016-08-18:
Hello Guys!
Andrew, I don't wanna waste your time so, here go my answers for your amazing comments.
Man, do not think that I am not enjoying the Carbon! You made a real toy here, the flow is exactly what I was expecting... so much fun! The Screen, the swing (yeah, the swing is working wonderfully)...the possibilities of layering sounds!

Thank you very much!

so, let's go!

"arp - Yes we plan to have more arp programs. As for other controls I guess it depends what sorts of controls you mean. I'm not against exposing more options but so far I didn't have any."
Well, recording midi arpeggios would be awesome. I spent some time thinking about the arpeggio idea... specially because this Carbon arpeggiator is solid, works for real. (I feel that all comments I made, even for me, are too early talking...). On my mind, if you gonna implement different  rhythmical possibilities for the arpeggiator... would be better recording MIDI arpeggios and reshape them with a increased "Pattern Generator".

"real-time recording of knobs - I'm not sure this fits into the current design as we really only record notes and CCs. It might be possible to have the song mode handle more parameters than currently, but I would have to hear specific things people want to do and we can see how that would work."
I can understand that. I just feel like, when I am playing around, I can make amazing variantions using the Carbon's knobs. Specially talking about gate time (rythm!)  and transpose... If we could record for example, different arppegio times in the same sequence... for example, switching from 1/16 to 1/132 and switching back. I'd like recording this kind of thing.

Why should I send these commands from another controller if I do have the knobs right there?
But I might been talking without knowledge. It's easy to see your points about the Carbon's  firmware structure.

"patterns affecting things other than the gate - Yes we can look into this. Again what would you prefer it be able to do? Right now the patterns really just hide events on a step which is fairly core to the design of CARBON. If you have other ideas of things it could do I would be interested in considering it."
Like Gate Time (again) or instead setting up a CC for single events you could use a pattern frame for setting up for more than one at the same time. We do have the CC chart, right? Why do not make pattern "masks" with it?
I love the patterns. Genius. You went deep in that so, you're the best person to say if we do need other pattern shapes.

"metronome volume control - The hardware is not capable of this, but there may be a software hack to allow at least some control over the volume. The metronome was a feature added fairly late in the design so we didn't have the ability to add a complete audio system to control it."
ok, it's just necessary in the first loop.

"notes per scene - Your desire is noted. The firmware currently has a way to compile it to support notes per scene instead of notes per song. But right now that would be a global firmware setting. A lot more work will be needed to allow it to work on a per-song basis because of how the memory is loaded and allocated, but it may be possible. I decided not to do this as I already explained. I see the desire for the feature, but it was too complicated for the first release." 
We can wait for that. Note that what we're saying is the oposite of what some people said about Carbon: "you can not make commercial music using this", which I think is tottaly wrong! People that work for real in the Business need this feature. 
**BUT I HAVE A QUESTION** "The firmware currently has a way to compile it to support notes per scene instead of notes per song." How can I do that? You mean, Can I record new notes in a channel without changing the previous scenes?

"clock glitch when changing scenes live - I'll have to look into this. I see no reason from having done extensive timing analysis of the running code. I believe this may be an obscure bug. Keep in mind that if you send out a lot of things on one MIDI port you might be running up against MIDI latency where changing scenes manually causes a lot of MIDI traffic. Because CARBON is quantized, many events end up getting sent out on the exact same clock tick which can cause the illusion of a glitch when it's actually just MIDI being slow. We are working on a feature that will automatically prioritize drum tracks ahead of voice tracks to try dealing with the limitations of 31250bps. :)"
I am practing and we gonna talk more about this. The Scenes are essencial for live jamming. I really do not like talking about other gear or other companies because I guess we can find solutions buy ourselves BUT, Scenes should have more options of playing like "sequential", "direct jump" or "start point".now, we just have direct jump.

"NRPN - Honestly I've never used NRPN for anything... it seems cumbersome. But I will have a look and see if it's possible."
what about "forget it"?

"Thank you so much for your comments! We are working on getting the code up on GitHub, and are also planning another round of development within the next few months to hopefully cover most of the requests that people have made so far."

So, I have two more "questions+sugestions":
1-Why not clock in? 
2- For example, If I want to sync the Carbon if another Carbon or, another sequencer as an MPC? Picture this: I am in a live act... I have my MPC loaded playing some sequences... in the other hand, playing in the same clock, I have another synced set up using Carbon, a drum machine and a synth... I wanna use them as separeted sources of music, like two turntables... how I can do that without some sort of clock in/out? Can Carbon trig another Sequencer for start playing?

Thank you Andrew... Of course I have more questions but now, let's work over the sequencer a way more. It's being an amazing time.

peace!

kilpatrickaudio, 2016-08-18:
Hi!

Just a quick follow up. All your points make sense and I will do my best to do as much as makes sense.

Wait, people said you cannot make commercial music using CARBON? Those people should keep in mind that CARBON isn't a songwriter, it's just a tool. And I think it's a pretty fun and flexible one. Sure it's not a clean slate like a DAW, but that's the point!

As for scene modes, this sounds interesting. The point of song mode was to allow for structuring a song, but I see that the real-time possibilities could be more interesting with different modes.

Why not clock in? What do you mean? CARBON supports syncing to a clock over USB or DIN MIDI.

Anyway, thank you so much for your detailed feedback. It really helps when people like you get engaged and share your experiences and wants. I fear that too many people these days just like to gripe on forums and never actually tell the designers what they want, so I appreciate it!

Andrew

closed 2016-08-18 by kilpatrickaudio