kilpatrickaudio/CARBON #9: Real-time Record Immediate Overdub

tripngroove, 2016-10-09:
Real-time record could be greatly improved by 2 changes, or possibly an alternate mode.

1. Allow notes to be added to the sequence immediately after pressing record, rather than on the next play-through of the loop.
2. Allow continuous 'overdubbing' until real-time record is manually deactivated.

The current "punch-in" real-time record behavior can be frustrating in a few scenarios in particular:

1. When working with a long or slow sequence, waiting until the next repetition to add notes can potentially be a very long wait.
2. When working with a short, fast sequence it can be challenging to place notes in the correct location because you need to hit record and immediately shift focus over to a midi input device (keyboard) and then play. The ability to hit record and then wait a few repetitions for the right moment to insert notes would be a big improvement.
3. Gradually building a sequence via real-time record can be frustrating because it requires constant re-arming, and because you can't re-arm while real-time record is active, you miss the ability to add notes every other repetition.

kilpatrickaudio, 2016-10-09:
Several others have requested very similar things via email, but I thank you for taking the time to break it down. Something along these lines is planned for the next release.

IrwinKwan, 2016-11-24:
I know that #20 was closed as a wontfix with a reference to #9, but I think that having a count-in is not mutually-exclusive with record immediate overdub. I would consider the following use cases:

User is programming tracks. The user's objective is to quickly enter track data. Current workflow (in v1.06) resembles:

1. User presses Record. (Recording is armed).
2. User presses play. (Sequencer begins; recording light blinks.)
3. User waits for one sequence to go around. (Pain point: user needs to wait for the sequencer, which was referenced by @tripngroove above in point 1.)
4. After 1 sequence, the user can now enter data.
5. In the situation that the user does not enter any data, record mode automatically turns off. (Pain point: the user needs to keep track whether the Carbon is going to record data or not, and needs to re-arm. This is @tripngroove's point 3 above).

Some possible designs to consider:

1. (Track recording) If I press "Record" before play, the Carbon would apply the count-in before starting the playback, allow the user to prepare.

2. a- (Real-time song recording) If I press "Record" while the Carbon is already playing, it can give a count-in before recording, rather than waiting for the next sequence. Such a count-in may be a way to address Point 2 that @tripngroove brought up.

2. b- (Begin Real-time recording when input received) As an alternative to count-in, if I press "record" while the system is playing, the system will wait until it receives input before it begins recording. If I happen to not enter anything in the current sequence, the record state will not change, and the system will be happy to wait for input.

3. (Record toggle) This is orthogonal to a count-in, but would address @tripngroove's point 3 above.

Personally I'd be a big fan of having a count-in so as soon as I press "Play" I can record after N beats without having to wait for the Carbon to cycle around the sequence once. I also wouldn't expect the Carbon to behave the same if I record while it's playing vs. if I record before starting playing.

kilpatrickaudio, 2017-01-15:
This requires major rework and will have to wait until another time. I'm going to mark this as WONTFIX but will in fact revisit it once I have time to get into it.

kilpatrickaudio, 2018-05-01:
We are working on a simple way to do this.

kilpatrickaudio, 2018-05-04:
Added in 1.14. A few points:

- Recording loops until stopped by pressing the recording button.
- Recording will "activate" the recorded notes at the end of each loop.
- It has always been possible to start a realtime recording without a lead-in by arming record and then starting the sequencer. We retained the lead-in function since once you're looping you can just keep recording. The lead-in function is useful in some cases and we didn't want to add more settings in the menus.

Please test this when the 1.14 release is out and report any weirdness as this required a lot of reworking of the recording code.

closed 2018-05-04 by kilpatrickaudio