kilpatrickaudio/CARBON #14: NRPN support for CCs

kilpatrickaudio, 2016-10-14:
A customer requested NRPN. Having never used gear that requires this, I don't have any way to test this. Please provide a use case and list some equipment that requires NRPN to perform the requested functions.

DavidProuty, 2016-12-01:
So NRPN would be very interesting. Roland gear is the first that comes to mind and much of the functionality is accessible through NRPN. All of the Vsynth's and the VP-9000 would be on my target list for use of that functionality ... However NRPN is complex and it would be a ton of work. Not to put this idea down but the time spent on this wouldn't be small and we would probably benefit more by that time being used elsewhere.

kilpatrickaudio, 2017-01-15:
Unfortunately NRPN is complicated to use correctly and I believe is beyond the scope of the basic CC support in CARBON. It could be simulated with multiple step sending the various CC incantation to send the NRPN command. Therefore if absolutely necessary it's already possible.

closed 2017-01-15 by kilpatrickaudio