kilpatrickaudio/CARBON #19: Insert or delete steps in a recording

kilpatrickaudio, 2016-10-14:
A user requested that notes could be inserted or deleted when in the step edit mode. This seems logical, but it would be necessary to figure out which button combination could do it without interfering with other modes.

DavidProuty, 2016-12-18:
Just a thought here. What if we made it so if while in step record mode if we held down the record button we could play a note or a chord on your keyboard controller and it would insert it at the location of the selected step.

kilpatrickaudio, 2016-12-19:
Hi David,

Step edit mode offers the ability to override the current step with new MIDI data. During step recording I don't think a feature to insert would work the way most people expect. I would likely want to add this into the step edit mode using a shifted key combination to slide over data from the currently selected step.

Does that sound like it would work well for you?

DavidProuty, 2016-12-20:
I sat down with Carbon for a couple hours last night while thinking about this and in use I would say we are already sitting in a sweet spot as of the 1.0.6 update. It might be better to leave the usage of those button's open for other features considering how well everything operates right now. If we slide data right and have a set amount of steps something must be lost at the end and I am not sure that makes sense considering we can so easily just replace a step.

kilpatrickaudio, 2016-12-21:
Okay, that sounds like a good idea. :+1:

closed 2017-01-15 by kilpatrickaudio