kilpatrickaudio/CARBON #23: Random / different motion types

kilpatrickaudio, 2016-10-14:
A user requested random / different motion types similar to how the K4815 Pattern Generator works. This seems like a good idea.

tomoe97, 2016-11-06:
I strongly support this request for different direction modes and motion types, such as random, ping pong, etc. I was really hoping Carbon could replace the K4815 Pattern Generator in my setup, but the Pattern Generator has additional motion types that are quite innovative, which are not replicated in Carbon. It would be great if some of the Pattern Generator's motion types could be ported over to Carbon.

kilpatrickaudio, 2017-01-15:
This would require too much change to the underlying architecture and so I have decided to omit this.

closed 2017-01-15 by kilpatrickaudio