kilpatrickaudio/CARBON #24: MAGIC should randomize gate lengths

kilpatrickaudio, 2016-10-14:
Currently the MAGIC function randomizes note values and velocities. A user requested that the lengths of the notes be randomized also.

IrwinKwan, 2016-11-13:
Would suggest that you consider if the feature would add delays as well for more rhythmic variations. For example, random insertions of half-step rests for a:

    1---2---3---4---
    N-----N---N-----

style of sequence would be a very cool addition.

Even more hardcore would be Euclidian rhythms, but I'll let you consider if that's what you're looking for in this.

kilpatrickaudio, 2017-01-15:
Due to time constrainst and to maintain product consistency we have decided to focus mainly on bug reports. Please consider forking the code if you wish to add major features such as this. Quality pull requests may be considered for the official release versions.

closed 2017-01-15 by kilpatrickaudio