kilpatrickaudio/CARBON #28: More arpeggiator programs

kilpatrickaudio, 2016-10-14:
A user requested more arpeggiator programs. Submissions would be greatly appreciated.

newsoner, 2016-10-17:
Yes, even for lower notes will be great

DavidProuty, 2016-10-26:
The pattern I would really like to see is Note-Order. IE (plays back in the order notes have been pressed)

kilpatrickaudio, 2016-10-26:
This is certainly possible. Thanks for the suggestion!

DavidProuty, 2016-10-27:
Here is a link to an interesting one page read on arp behavior, just something to maybe help your inner muse.  http://lauterzeit.com/arp_lfo_seq_sync/

kilpatrickaudio, 2017-01-15:
I will attempt to write some additional arp patterns in a future release.

hamildad, 2017-05-09:
Took a look at Arpeggionome this morning.

In addition to the usual, they had:

Thumb up 1-2-1-3-1-4-1-5-1-6 etc.
Pinkie down 6-5-6-4-6-3-6-2-6-1 etc.
Diverge    3-4-2-5-1-6 or 4-3-5-2-6-1
Converge 6-1-5-2-4-3 or 1-6-2-5-3-4
Diverge&Converge 3-4-2-5-1-6-6-1-5-2-4-3
Converge&Diverge 1-6-2-5-3-4-4-3-5-2-6-1

hope this helps!

kilpatrickaudio, 2018-05-01:
These are good suggestions, thank you.

kilpatrickaudio, 2018-05-05:
Added more arps in 1.14.

- note order
- up/down with no repeat of top and bottom notes
- up, lowest note between each note
- down, highest note between each note
- repeat whatever is held down

closed 2018-05-05 by kilpatrickaudio