kilpatrickaudio/CARBON #29: More tonality scale types

kilpatrickaudio, 2016-10-14:
A user requested more types of scales in the TONALITY filter module. Submissions would be greatly appreciated.

DavidProuty, 2016-11-01:
We do have some of these currently however it would be great to have more.
**major (ionian),dorian, phrygian, lydian, mixolydian, aeolian (natural), locrian, harmonic minor, melodic minor, whole tone, major pentatatonic, pentatonic var, pentatonic 1, pentatonic 2, rel. minor pentatonic, egyptian, shomyo, new pentatonic, pentatonic 3,pelog, balinese, japanese, balinese 2, aug. hexatonic, hexatonic 1, native american, mystic hexatonic, hexatonic 2, hexatonic 3, blues 1, hungarian major, heptatonic 1, leading whole tone, lydian minor, overtone, hindu, locrian major, locrian mod., algerian, hungarian gypsy, blues 2, heptatonic 2, enigmatic, heptatonic 3, heptatonic 4, heptatonic 5, double harmonic, phrygian minor, persian, heptatonic 6, neapolitan major, javanese, neapolitan minor, xotic, major bebop, major dim. fifth, messiaen m.2, messiaen M.4, calliope, octatonic 1, octatonic 2, messaien m.1, spanish, chromatic

I'm not sure if this would save any work but this link is to a Synthedit Sem file http://www.xoxos.net/sem/voltstoscale.zip This sem can be opened in Synthedit, once that is done the code for it can be output by right clicking the V2Scales module and the selecting the Build Code Skeleton.  It exports to a folder as C++. This Sem is volts to scale and contains all the above listed scales. I believe Synthedit works on every 0.5 is a semitone but it could be 1volt so you would have to check.

kilpatrickaudio, 2016-11-01:
We have to be very careful about licensing for external code or tables brought in. Check out the tables/scale_tables.h file to see how the tables are currently made. The quantizer automatically fits any note to the chose scale so adding more is easy. If anyone wants to supply more scale types and double-check them against music theory it would be great. :)

Jumbuktu, 2016-12-22:
Just worked out I can upload files here. Here is my text file edit to add the obvious scales to the tables. 

[Scales.txt](https://github.com/kilpatrickaudio/CARBON/files/668316/Scales.txt)

kilpatrickaudio, 2017-01-03:
Thank you! I will integrate it soon.

Jumbuktu, 2017-01-03:
Ok. I just noticed that there is other code outside these tables that affects scales. You might need to review it as well, since it includes reference to specific scale names.

kilpatrickaudio, 2017-01-15:
Some more scales were sent to us by Jumbuktu... thank you! I will attempt to integrate these in the next update.

Jumbuktu, 2017-07-12:
Added a few more: 
[Scales2.txt](https://github.com/kilpatrickaudio/CARBON/files/1140577/Scales2.txt)
These are all in the public domain.

On a related topic, I suspect that some of the Wendy Carlos scales can be  created on Carbon by (ab)using the octave scaling facility.

kilpatrickaudio, 2017-12-14:
There are 8 new scales in ver. 1.12. Some submitted scales are non-equal tempered according to my research and these were not included due to discrepancies between interpretations. But hopefully this will add more flavour!

closed 2017-12-14 by kilpatrickaudio