kilpatrickaudio/CARBON #30: More pattern types

kilpatrickaudio, 2016-10-14:
A user requested more pattern types. Currently the 32 patterns (with some minor tweaks) from the K4815 Pattern Generator are supplied. Submissions would be greatly appreciated!

Jumbuktu, 2017-01-03:
I suggested on another issue that it would be nice to be able to hit a button and randomise the cells of the current pattern. If there was an option to save over the current pattern, this would give an endless supply of possible patterns, without someone having to try to guess what might be useful.

boerup, 2017-01-12:
The ability to add your own patterns would solve this issue.

Jumbuktu, 2017-01-13:
I agree with boerup that we need a user pattern editor, but i am not sure that solves the original issue.  The original issue asked for 'more pattern types', presumably more than 32.

There seem to be a bunch of separate enhancement threads for patterns, and i think it would make sense for Andrew to consolidate them into a single one.

The 'issues' seem to be:

1. Can we have more pattern memory slots?
2. What should go in those slots in the default firmware? [This seems to be the help Andrew is asking for in this thread]
3. Can we have a way of editing and saving user patterns?  This could be in an external computer-based editor (but then we would need a way to get them in) or (better) within the firmware.  The patterns are pretty simple - just a 8x8 binary array - you would just need a way to navigate through the matrix (one of the knobs) and a way to switch the selected cell on or off (the other knob).  As boerup says, this would eliminate the need for an answer to 2 - users would determine what is useful to them.
4.  My idea of a randomise function to fill the cells of the current pattern.  This would be easy to implement and would provide a quick way for generating experimental patterns.
5. Can we have patterns affecting other parameters, such as velocity (this would work like an accent feature on a drum machine, and would be really useful IMO).
6. Could pattern data be expanded from binary (on/off) to a numerical range, which could then be used to control velocity or some other parameter (this was one of my suggestions)?

These are all attempts to make the pattern feature a bit more useful.  In general, the patterns seem a bit out of place in Carbon at the moment.  The Carbon hardware and connectivity give the impression of a solid, serious sequencing control centre for a mixed MIDI and CV/Gate system.  However, the patterns as they stand take up most of the screen real estate and seem a bit like a 'toy' feature with limited uses.  This might be Ok in a single experimental Eurorack module, but it seems like wasted resources for Carbon.  Some of these enhancement suggestions might help to improve the situation.

PS one thing that would help me take the patterns a bit more seriously would be to ditch the silly 'smiley face' pattern and replace it with something like a series of Euclidean rhythm patterns.

kilpatrickaudio, 2017-01-13:
Thanks for the feedback. Since the pattern, scale, arp stuff seems to be what people want more of, those will be on the top of the list for the next update. (probably February) As for other similar threads, Github doesn't allow issues to be deleted, so it's hard to "merge" stuff. But for now we'll go with this one as the main issue for discussing anything pattern related as that seems like a good idea.

To comment on the previous post:
1. there can be more patterns - 32 was just a default number but there is additional memory space
2. the defaults can be changed but mostly I want to keep the first ones as based on the "legacy" ones from the K4815 - new ones can be created as needed
3. the KPatternEdit software that we made for the K4815 can be changed to support CARBON - in fact in the meantime if anyone wants to use it to make some new patterns I would be happy to consider adding them in the "new" slots
4. randomize is a good idea - it would be easy to implement using the MAGIC feature
5. the pattern feature working as a gate pattern is fairly heavily "baked" into the current operation - making it do other things is a possibility but would require a lot of reworking since a lot of things assume / depend on it - this requires some thought
6. as for continuous pattern stuff, I'm just going to say no on that for now - controlling and visualizing this is very hard and it's not really the essence of what CARBON is, or what it's based upon

It's great that there are so many creative folks thinking of new concepts and uses for CARBON. We made the code open so people could fork it and remix things as they feel like. But keep in mind that as a product, we (Kilpatrick Audio) are aiming to stick with what is there now and making incremental improvements based on feedback.

Jumbuktu, 2017-01-13:
Here is a Excle spreadsheet with a few ideas for additional patterns.  Some of these replicate existing patterns but hopefully there are some new ones as well.

[Pattern_generator.xlsx](https://github.com/kilpatrickaudio/CARBON/files/703553/Pattern_generator.xlsx)

kilpatrickaudio, 2017-01-15:
I will attempt to add more baked-in types. Not sure if uploading will work for a while as I need an app for this.

Jumbuktu, 2017-01-17:
I forgot to mention ... I know you are not doing enhancements, but this is not really an enhancement.  In the manual it refers to 'the highest numbered pattern'.  However, the Patterns don't have numbers, unless I have just missed them.  It would really help to know where we are when scrolling through, especially if you add more user Patterns.

kilpatrickaudio, 2017-01-17:
Okay, we'll look into this and see if there is a spot to display the current pattern as a number.

kilpatrickaudio, 2018-05-07:
Added in version 1.14. The number of patterns is still 32 (including "as-recorded") as stated earlier. However the system patterns (which are normally loaded from config flash at startup) can be changed. These are saved back globally into the config store. Usage works like this:

- SHIFT + STEP EDIT goes into pattern edit mode
- use PATTER to select a pattern to edit
- use SELECT and VALUE to edit the pattern steps - changed are stored automatically
- to recall a factory ROM pattern to a slot press SHIFT+ RECORD [CLEAR] to restore it

closed 2018-05-07 by kilpatrickaudio