kilpatrickaudio/CARBON #31: Patterns should be able to affect non-gate things

kilpatrickaudio, 2016-10-14:
A user requested that patterns be able to control things other than whether events on a step are performed. (note / CC sent) Please help to clarify how this would work in the context of CARBON?

Currently I'm tempted to mark this as invalid / wontfix but am open to ideas! :)

tripngroove, 2016-11-17:
Could be cool to allow patterns to affect the on/off states of:
- arpeggiator
- portamento
- transposition (this one could be particularly interesting)

Also, the ability for a pattern to apply offsets to: 
- entered CC values
- note lengths
...sort of turns patterns into complex LFOs w/ mod matrix.

But if you have the above, you inevitably want to be able to assign one pattern to notes, a different pattern to arp state, etc...

Jumbuktu, 2016-12-22:
At the moment, pattern cells are either 100% on or 100% off. Why not let each cell be any number between 0 and 127, and apply the number to velocity? That would make the patterns into something like a groove template (e. g.  100, 60 ,75 ,60). Zero could mean the same as it does now (i.e. no gate or trigger) and the cell colour could reflect the value. 

Also, it would be nice to hit a button to randomise the cells of the current pattern. If it turned out to be a good one there could be an option to save it over one of the existing patterns.

kilpatrickaudio, 2017-01-15:
Due to time constrainst and to maintain product consistency we have decided to focus mainly on bug reports. Please consider forking the code if you wish to add major features such as this. Quality pull requests may be considered for the official release versions

closed 2017-01-15 by kilpatrickaudio