kilpatrickaudio/CARBON #37: Transpose by Octave

lineofcontrol, 2016-11-02:
Hi Andrew!

I rarely transpose a playing pattern by a few semitones but almost always like transposing patterns by an octave up/down for a bit of tonal variety. 

Would it be possible to choose in a settings menu?  :

Transpose types:

1) Semitones (default)
2) Octaves (12 semitones)
3) Thirds (3 semitones)
4) Fifths (5 semitones)
5) Sevenths (7 semitones)

Thanks so very much for a wonderful product and all your help above and beyond the call of duty.

Sunil S. 

P.s. I just programmed the exact same pattern in my Carbon and in my Audio damage Seq1 and they sounded rather different in terms of timing...the one from the Carbon just sounded more musical and tight. Pretty interesting difference!

lineofcontrol, 2016-12-22:
Any thoughts on this???

Jumbuktu, 2016-12-22:
I am new to Carbon. I tend to use transposing to create variations, such as up a minor third. I can see how a menu might be useful, but I would place higher priority on a few other enhancements.

Jumbuktu, 2016-12-22:
PS 3 semitones is only a minor third, and a fifth would actually be 7 semitones. 5 semitones would be a fourth.

lineofcontrol, 2016-12-22:
Ooooops. Sorry.  My lack of actual musical theory knowledge is showing!

kilpatrickaudio, 2017-01-15:
Due to time constrainst and to maintain product consistency we have decided to focus mainly on bug reports. Please consider forking the code if you wish to add major features such as this. Quality pull requests may be considered for the official release versions

closed 2017-01-15 by kilpatrickaudio