kilpatrickaudio/CARBON #39: Damper pedal should cause ARP to hold

kilpatrickaudio, 2016-11-03:


DavidProuty, 2016-11-13:
Agreed. It is very useful to be able to hold an arp.

Jumbuktu, 2016-12-22:
Do you mean the Sustain pedal? 

Presumably this would act as a toggle to turn hold on and off, rather than having to hold down the pedal to keep the Arp running?

IrwinKwan, 2017-04-30:
I would actually like some way to allow for a free-running ARP while the system is playing regardless of damper pedal or not. Maybe an option in the ARP menu to specify if it's "one-shot" or "hold". I'm actually surprised that "hold" isn't the default.

For example, in Unlimited, f I set the pattern step to 1/1 and an arp step to 1/16 and have C E G then it would be marvelous to have C E G run 16 notes until it moves to the next step.

kilpatrickaudio, 2017-05-01:
This is a good idea. I'll try to include it in the next release.