kilpatrickaudio/CARBON #42: Song 5 on my Carbon seems to have random problems

IrwinKwan, 2016-11-11:
I have NO idea how to indicate to others how to reproduce this, but on my unit Song 5  seems to have all kinds of random problems that so far I haven't been able to reproduce on other tracks. Here are some of the issues I've experienced:

1. When recording while the sequencer is running, at the end of the sequence, the run will stop, the settings will be erased, and my tracks will get odd settings like "Scale: Scene 1 Track".

2. Once Scale: Scene 1 Track appears, the sequencer tracks will run in odd directions. Some reverse, some forward. 

3. Recording further information to the track can't be accomplished.

4. I'm unable to clear some of the tracks.

The symptoms seem to indicate some kind of problem with the song. Is there some way to restore that song to default or similar and avoid these problems?

kilpatrickaudio, 2016-11-11:
Please email us directly: andrew@kilpatrickaudio.com This sounds like some kind of issue with your hardware and I would like to help you debug it one on one.

closed 2016-11-11 by kilpatrickaudio

kilpatrickaudio, 2016-11-12:
This is fixed and will be part of 1.06 release.