kilpatrickaudio/CARBON #44: Change MIDI input to omni mode

kilpatrickaudio, 2016-11-11:
MIDI input is now omni so any channel can trigger notes or CCs while recording or playing. Notes are still output on their corresponding channels but any channel (1-16) can be used. Some controller devices come with preset channel numbers which cannot be changed.

kilpatrickaudio, 2016-11-11:
Fixed.

closed 2016-11-11 by kilpatrickaudio

IrwinKwan, 2016-11-13:
I am hopeful that this will resolve the USB input issue with Multidimensional Polyphonic Expression controllers like the Linnstrument, which try to get around limitations of MIDI by using multiple MIDI channels to send MIDI CC data. Thanks a lot,

DavidProuty, 2016-11-30:
Excellent. This will make things better on many of my devices. Any idea on a release date yet?

kilpatrickaudio, 2016-11-30:
Try version 1.06... it should be there now.

DavidProuty, 2016-11-30:
Oh that's great! I was just there before I went to Github to check on
progress. Looked right at the version 1.0.2 and thought to myself well it's
not released yet. It was a long day here but really no excuse. Thanks for
sending me back over there. ;)
[image: Inline image 1]