kilpatrickaudio/CARBON #47: Invalidate BPM display when using external clock

kilpatrickaudio, 2016-11-20:
A user requested that instead of greying out the BPM display when using an external clock, that it display something else like "---" to indicate that the number displayed is not valid in the current mode.

flopass, 2017-03-20:
It would be great if it could display the current external tempo as it changes in real time. Possible?

kilpatrickaudio, 2017-03-20:
I've definitely considered this, and it's fairly trivial to do. But in my travels I've read about a lot of other products and software that do this, and the number one complaint from users is that the display is not accurate or it jumps around, as is the case when locking onto a clock from another system. Playback will be fine, but the display might not perfectly match the one on the sending side.

Thoughts? Maybe there could be an option to disable it...

flopass, 2017-03-20:
My experience comes from the snazzy way I have my Elektron Analog Keys (master) communicating with the op-1 (slave) and when I change the tempo in real time it adjusts so I can see it on the op-1. It's certainly not a necessity, and sometimes it sways .5 from the tempo but it helps me know that not only is it communicating, but it's following the new tempo. Basically it mostly serves to make me feel better. Once I've changed tempo in real time I usually hard restart everything in the chain to get everything synced back up properly, but it is nice to know it's trying. Those are my 2 cents. Thanks for listening!

-Raymie

kilpatrickaudio, 2018-05-01:
This is done now.

closed 2018-05-01 by kilpatrickaudio