kilpatrickaudio/CARBON #49: How to UPDATE FIRMWARE with MAC ?

Roisner, 2016-11-24:
Hi,

I don't have PC and cannot run DfuSe Demo Software.
Is there another way?

Thx

IrwinKwan, 2016-11-24:
Hi Roisner, check this out. https://irwinkwan.com/2016/11/23/upgrading-the-kilpatrick-carbon-firmware/

kilpatrickaudio, 2016-11-24:
We will publish official OS X instructions once we test them. It is also possible to use the dfu-util program on Linux.

closed 2016-11-24 by kilpatrickaudio

Roisner, 2016-12-06:
Hey Irwin , thx you for your help ! I try but without success :( 
Will try with a PC of a friend !

IrwinKwan, 2016-12-07:
What problem did you encounter? In the instructions I've written I did skip over some steps like setting up Xcode and installing HomeBrew, so if you are running into those problems I'd like to see if I can provide some more assistance in that way.

Roisner, 2016-12-09:
The problem was that on TERMINAL , CARBON does not appear .
Scheinderladen shop in Berlin did it for me and told me that it was because of the drivers when his on UPDATE MODE.