kilpatrickaudio/CARBON #54: Play live while track is running

boerup, 2017-01-09:
Playing live while the same track plays would be very useful. 

I often record small, simple sequences and want to improvise on top of these. This is not possible at the moment and it degrades the live use of Carbon in my opinion. 

Br, AB

DavidProuty, 2017-01-15:
Yes and it would also be nice to do live transpose with your left hand while playing melody's & chords with your right hand. We really should be able to create a split point that allows for this. Also the ability to transpose the left and right independently and set the midi in and out channels for left & right. I know that I have brought this up before but the more I use Carbon the more it becomes apparent this is a major weakness of what we have currently.

kilpatrickaudio, 2017-01-15:
Due to time constrainst and to maintain product consistency we have decided to focus mainly on bug reports. Please consider forking the code if you wish to add major features such as this. Quality pull requests may be considered for the official release versions

closed 2017-01-15 by kilpatrickaudio