kilpatrickaudio/CARBON #55: USB 3 on USB host causes lockup of that port

hamildad, 2017-01-10:
When I plug in a USB 3 connection (from ROLI Blocks) into USB host, it locks up that port and I cannot use it for USB MIDI input, if I replace with USB1 or 2, the ports stays locked until I power cycle the unit.

kilpatrickaudio, 2017-01-10:
This is likely a bug in ST's USB driver system. Our custom class drivers simply implement the USB MIDI class according to the spec. But the underlying USB host code doesn't seem to fail very gracefully.

I'll spend some time in my next round of changes to see if we can at least keep the USB subsystem from hanging when a bad device is detected. But it's unknown at this point whether USB3 can be supported with the hardware that exists in CARBON.

kilpatrickaudio, 2017-04-05:
I'm going to close this for now since I do not have any USB 3 devices to test with, and USB 3 cannot be supported by the hardware. If someone wants to lend me some ROLI gear to test with I can verify the actual problem with our USB analyzer.

closed 2017-04-05 by kilpatrickaudio

hamildad, 2017-04-22:
updated both Carbon and Roli Blocks and same cable and unit does not lock up system anymore. in fact I can use a Blocks to note input into Carbon...  mark as fixed if you want.

@

kilpatrickaudio, 2017-04-25:
Thanks for the update! This is good to hear.