kilpatrickaudio/CARBON #58: USB host doesn't always work

boerup, 2017-01-12:
I have an Akai LPK25 keyboard. 

Sometimes, when I plug it in Carbon, it works. 
Sometimes, when I plug it in Carbon, it doesn't work. 

It works flawlessly every time I plug it into my computer. 

What's going on? I am going on tour soon, and would like this to work!

Br, AB

kilpatrickaudio, 2017-01-12:
This is an issue we haven't been able to track down yet. In our experience once the USB host port enumerates a device properly, it continues to work. Occasionally the device fails to enumerate.

Unfortunately the USB host stack on the STM32 (the processor that CARBON uses) seems a bit fragile. It's something that we're hoping to get to the bottom of, but involves a lot of code that we didn't write and isn't very well documented.

boerup, 2017-01-17:
So, what do you suggest I do while I wait for the fragile STM32 to be inspected? If I want to use my Carbon, that is...

kilpatrickaudio, 2017-01-17:
In my experience once the host is enumerated properly it continues to work correctly. I've seen this same behaviour on many devices and computers too... where sometimes you plug in a device and it doesn't respond, but once it's working it continues to work.

If you want to avoid the USB host until a 100% solid fix is available, there is always the DIN MIDI IN. But we've done many tests and demos with the USB host port with no issues.