kilpatrickaudio/CARBON #59: More scenes, please

Jumbuktu, 2017-01-15:
Aside from their use in live flipping between settings, I find Scenes of most use in setting up a Song.  However, only 6 Scenes is pretty restrictive in this context.

Technically, it is probably not all that difficult to just allocate more space for additional Scenes.  I suspect the major reason for not adding them would be to avoid complicating the operating system.

From the point of view of Song editing, a bigger Scene list would be no more difficult to use than the current 6.  The Scene to add to each Song step is just another parameter anyway.

So the main area that would need to be thought through would be the interface for selecting, saving, copying and loading Scenes.  There are plenty of different ways of doing this - here is my suggestion based on just adding another 6 Scenes:

1.  Select one of Scenes 1-6: Leave as is (hold down Scene button and select one of the track buttons)
2.  Select Scenes 7-12: Hold down Scene button plus Shift button and select one of the track buttons [Yes, I know this is currently used for copying Scenes, but it seems more logical that the alternate bank of 6 Scenes would be accessed using Shift]. 
3. Copy current Scene to one of Scenes 1-6:  Hold down Scene, Load/Save and select one of the track buttons [same as current but using Load/Save instead of Shift].
4.  Copy current Scene to one of Scenes 7-12: Hold down Scene, Run/Stop and select one of the track buttons [same as for 1-6, but use the next button to access the second bank of Scenes].

This would be reasonably straightforward.  It only adds 6 more Scenes, but 12 in total would be enough to be able to make proper use of the ability to shift position in several sequences at once and build variations plus complete new sections.

Any users who think this would be useful, please add your support.

kilpatrickaudio, 2017-01-15:
Due to time constrainst and to maintain product consistency we have decided to only focus on bug reports. Please consider forking the code if you wish to add features beyond those which currently exist in CARBON.

kilpatrickaudio, 2018-05-07:
I'm going to reopen this one based on Jumbuktu's request. It will have to wait for the next maintenance release since 1.14 is almost done and we want to get it out. (it already has a ton of stuff)

We will investigate the SHIFT+SCENE idea as suggested for getting access to 6 more scenes. Not sure how we will do scene copying, but will obviously have to keep that somehow.