kilpatrickaudio/CARBON #60: Song list end point action

Jumbuktu, 2017-01-24:
I know Andrew is concentrating on bug fixes rather than enhancements for the time being.  I will just put this here for future reference.  It is a fairly simple, relatively minor change.

At the moment, when Carbon reaches the end of a Song List , it just reverts to playing the current Scene over and over.  This is Ok in a Live situation or where you want to just fade out.  However, there ought to be a few alternative ways of finishing a Song:

1. The sequencer stops playing and the Song pointer is set back to the start of the Song list (ready to start if Play is pressed again).

2. The sequencer keeps playing but goes back to the start of the Song list and keeps playing the Song over and over ad infinitum.

3. The sequencer keeps playing but goes back to step n of the Song list and keeps playing from step n to the end of the Song over and over ad infinitum.

4. The sequencer keeps playing but stays on the last Scene of the Song list.

5. The sequencer keeps playing, but goes back to Scene n.

I think 4 is the current default.  Personally, I think 1 or 2 would be a better default, but it is not a big issue.

Incidentally, I find the use of the term 'Song' a bit confusing.  The entity that is called a 'Song' in the Carbon manual might be better called a 'Sequence' to distinguish it from the 'Song list', which is really what we all know as a 'Song' from common usage in drum machines and groove boxes.  It's not a big deal once you understand the terminology.

kilpatrickaudio, 2017-04-05:
This is a good idea. I think we will implement #1 as it makes the most sense.

overthereinabox, 2017-05-16:
Andrew, would you also be able to consider song chaining?