kilpatrickaudio/CARBON #63: MIDI external clock needs to support rapidly starting clock

kilpatrickaudio, 2017-02-13:
Version 1.07 supports excellent clock locking for external MIDI clock, but assumes that the clock is always running even when the transport in the parent application is stopped.

Some programs (Ableton) don't sent clock when stopped and have no option to turn this on. We need a way to quickly sync with a clock that starts as soon as playback starts, and get up to speed without glitching.

DavidProuty, 2017-02-15:
Not sure if this is the same thing but in Cockos Reaper when I start transport Carbon will start playback once or twice then on the third start or so the RUN/STOP button LED lights and the upper top left RUN indicator turns green but Carbon doesn't start playback. It will stay stuck for multiple transport start/stop's and then eventually start running again.

kilpatrickaudio, 2017-02-15:
Yes, this sounds like the same thing. You currently need to have your software support a continuously running clock for CARBON to sync correctly.

DavidProuty, 2017-02-15:
I assume since this is on the Bug list it will be fixed. It's safe to reload 1.06 back for the time being correct?

kilpatrickaudio, 2017-02-15:
Versions prior to the latest (1.07) do not support proper external clock syncing and can drift out of sync over time. 

Version 1.07 addresses this with a proper locking routine that stays in sync with the external clock. But it currently relies on the MIDI standard where the clock runs even when transport is stopped. Apparently some newer programs don't offer this option, and thus will not currently work correctly with CARBON.

The next time we can schedule maintenance work on this code this is our top priority. We do not have an ETA but this issue will be closed when it's done and tested.

kilpatrickaudio, 2017-04-05:
This is fixed in commit 90bca8c and will be tested and released in ver. 1.08.

closed 2017-04-05 by kilpatrickaudio