kilpatrickaudio/CARBON #66: Clock

DavidProuty, 2017-04-10:
Carbons clock is very stable on its own and that's great but linked to anything else the clock is currently all over the place. It would be nice if we had an option to just except the start/stop message from an external device so that carbon could start with those midi sources while using Carbons internal clock due to its accuracy. IE (Use Carbons internal clock but except start/stop messages from external gear). This would save a lot of headaches. I'm going to reload 1.0.6

kilpatrickaudio, 2017-04-10:
My responses below are based on version 1.08, which I assume you are using.

The CARBON clock is designed to respond to MIDI start, continue and stop messages even when set to INT mode. Did this not work for you?

As for the clock drifting, the drift is automatically corrected on the fly as the external ticks are compared with internal timing. As long as the external clock is stable, it should work very well.

DavidProuty, 2017-04-10:
Yes I am using the 1.0.8 version and when Carbon is set to Midi Clock Source Int it is not responding to start, continue and stop messages from Cockos Reaper nor Ableton Live. I haven't tried any other DAW's. I can do some trouble shooting if you like.

kilpatrickaudio, 2017-04-10:
The clock input must be set to an external port to respond to start, continue or stop messages on that port. Please consider using the MIDI remote control functions which offer an alternate way to control the transport while also allowing CARBON to ignore external clocks.

closed 2017-04-13 by kilpatrickaudio