kilpatrickaudio/CARBON #68: Pitch Bend Record

MDTF, 2017-04-17:
It would be great if the Carbon can record and play the pitch bend/modulation wheel of a midi keyboard. The pitch bend is often an essential part of the melody. Moreover, for many synths, it can be remapped to other parameters, so this basically amounts to recording one continuous parameter.

(Something slightly similar was also mentioned in issues 8 and 57)

(Possibly this requires too much memory? The precision is however not extremely important)

kilpatrickaudio, 2017-04-25:
This is a good idea... we will look into it as an option. The only issue is that it would be quantized so I'm not sure it would really give the effect that you're after.

MDTF, 2017-05-03:
I was thinking about maybe a linear or quadratic interpolation between the quantized values. Of course one looses precision in the process, but that shouldn't matter much, given the way the mod wheel is usually used musically.

kilpatrickaudio, 2017-05-03:
Alright, we'll look into this.