kilpatrickaudio/CARBON #69: Can't play live while pattern is playing

boerup, 2017-04-20:
I don't get it: 

Why won't a track play while I play live on the same channel? 
It really ruins the live use of the Carbon. 

I record something on one track - routing it to, say, a piano sound. 
Now, playing live, I want to add small notes to this newly recorded line. 
But I can't - because when I press rec, the track is muted. 

This really doesn't make sense, if the Carbon is supposed to be for creative, improvised use in a live situation. Couldn't this be resolved with a firmware update?

Br, B

kilpatrickaudio, 2017-04-20:
The purpose of LIVE mode is to take over a track and play from a keyboard instead of the sequence. This is intended behaviour and is not a bug.

If you wish to have notes from a track and live output at the same time, perhaps consider setting up another track with the same MIDI output settings as the sequencer track. This should allow both live and sequencer playback at the same time.

We will consider your request as an option for the future, however.

closed 2017-04-20 by kilpatrickaudio

boerup, 2017-04-21:
Just imagine: You play a small pattern live, it keeps running, you add small details to it whenever you want, and the track evolves creatively. Really, that the track stops when you want to add stuff to it, is a show stopper if you want to use Carbon in live situations. Without this feature, it's just a player for pre-made sequences = boring music....

Please, please, please add this option!