kilpatrickaudio/CARBON #72: Bias track info not stored within Scene, but appears to be Global.

hamildad, 2017-05-08:
Firmware 1.8

it seems impossible to have some scenes using a bias track and other scenes not. So it seems that Bias is global to the song, where the Carbon Manual states.

> The following parameters are stored in a scene for each track:
> bias track settings

So it should be possible to have Bias tracks for a single scene within a song, and use the (bias) track for audio on a different Scene.

kilpatrickaudio, 2017-05-08:
I'm testing this and it looks like the bias track is being switched per scene. Please offer steps to reproduce your problem.

hamildad, 2017-05-10:
I had a look at this and it related to issue #71 a lot if information is not getting fed back to the GUI when you change scenes.

I was able to change scenes and have the GUI stay on the previous bias/direction and tonality setting, whilst going to the menu showed the scene was behaving correctly.

This should be easy to replicate by creating a number of scenes with different direction/bias/tonality settings and switch through them. comparing screen against behaviour.

kilpatrickaudio, 2017-05-10:
This should be fixed by the changes made regarding the direction setting. The entire track data is refreshed on the screen when the scene changes.

closed 2017-05-10 by kilpatrickaudio