kilpatrickaudio/CARBON #80: Drum mode little enhancement for tracker user ;)

arcangeli, 2017-07-01:
Hi,

What can be really handy in drum track, is a possibility to record (in step edit mode) from keyboard directly on the selected event. Actually, only the first event is recorded if there is only one key typed on the keyboard.
It is possible to do this manually (add only an event on the 3rd event) but not from external keyboard.
In my mind (coming from tracker some years ago):
Track 1 is Drum. Event one is always kick. Event two is snare.
Actually, if i type only one note on the keyboard, carbon always record it on the first event. Not on the selected event.
Is it possible to change this behavior, please? 

Thanks beforehand
Bruno

kilpatrickaudio, 2017-07-03:
Let me think about how this could work... perhaps in DRUM mode the insert / replace behaviour could be slightly different as you request. Leave it with me and we'll see what we can do.

arcangeli, 2017-07-04:
I'll try to be more consise. English is not my native language ;)
On drum track in step edit mode. I want to add only one event at a time on the selected event (shift+select). Add another event on the same track on another event number.
It's possible to do that manually on the Carbon but not with a controller as any key/knob press erase all event.
This is good with chord on voice track but not on drum track. Many times, i want to add/replace just one event on a step.

kilpatrickaudio, 2017-07-04:
I see what you mean. Perhaps we can make it so when a single event is selected only that slot will be filled with a new MIDI event and others are left the same as before. Would that work for you? If so, that's a fairly simple change that makes sense.

arcangeli, 2017-07-04:
Exactly :)
It would be excellent.

kilpatrickaudio, 2018-05-01:
I've decided to pursue this as part of the STEP EDIT mode by making a single slot record when the user has a slot specifically selected. Currently this is only used for editing, but it would make sense to have the ability to populate the slot of your choice as well. This should allow the kind of editing you are describing.

arcangeli, 2018-05-01:
Awesome! Thanks very much.
I <3 you ;)

kilpatrickaudio, 2018-05-04:
Tracks set to Drum mode will now not clear a step in step edit mode. Only notes of the same pitch will be replaced with new ones. To clear a step or note on a step requires manual clearing.

closed 2018-05-04 by kilpatrickaudio