kilpatrickaudio/CARBON #83: Quantise Scene changes

hamildad, 2017-07-26:
I'd really like it if you could set scene changes to be quantised to the end of a section.

the menu option would be in clock menu and offer quantised or not quantised, 
Then whatever track you were on, would delay any scene changes to the end of that sequence,
This will stop you having to 'punch in' scene changes and then get it just a bit off when playing live.

sometimes for a fast clocked rhythm sequences I punch out on the 31st beat and the whole thing loses a measure of time.

kilpatrickaudio, 2017-07-27:
Currently all scene changes are synchronized to the beat. It's not possible to synchronize the changes to the end of an arbitrary section because this would only work on a single track. We could make it possible to choose a multiple of some number of beats so it could happen every 4th beat or something, but finer resolution causes problems and simply won't be what you want. (this was the original behaviour during development)

hamildad, 2017-07-27:
finer resolution would be worse and I want to know I will change scene in the gap between say 16 -> 1.  its the possibility that you might change scene in the gap between 15 beat->16 beat that I want to avoid.

but close this if thats not possible.

kilpatrickaudio, 2018-05-04:
Fixed in 1.14. Song now has a mode accessible in the CLOCK menu which selected scene change sync of: Beat or Track 1 End

closed 2018-05-04 by kilpatrickaudio