kilpatrickaudio/CARBON #84: mute on bias tracks

hamildad, 2017-07-26:
I would like the ability to use bias tracks to mute sequences, so I dont have to use up 32 steps for a 16 bar sequence that I want to play every other measure... 

either:

no note in a bias track mutes it, so four bars of F would have to be F,F,F,F rather than F - - -  as it currently is.

or use something like C0 as the mute trigger.

both of them might be confusing to the new user, but would really open up the Carbon to non continuously repeating patterns.....

kilpatrickaudio, 2017-07-27:
Hmm, that's interesting. I think this could be done as a mode... so if a track is set to be driven from a bias track it could be set as transpose, mute, or both. Would that make sense? The default will be "transpose" so it works as originally designed.

hamildad, 2017-07-27:
keep transpose as the default, but a way to mute via the bias track would allow me to make better use of the 64 steps within a pattern by having Bias mute and transpose depending on what was selected.

The issue is user confusion where tracks are not playing due to a mute state on a different track, but you have that with transpose, so I guess its making it clear in the GUI.

kilpatrickaudio, 2018-05-01:
I think we can implement both mute and normal. I agree that the bias track should always bias. If you put a C it will not transpose, if you put some other note, it will. If the Mute function is also selected (bias track would be shown as <3M instead of <3 on the track status display it will do the mute function as well as the transposition.