kilpatrickaudio/CARBON #85: Incremental changes to patterns using Magic

hamildad, 2017-07-26:
a setting in Tonality menu where Magic will either change all the notes or a single note per press.

So that multiple presses of Magic will slowly evolve a sequence over time....

you would need to have it change a note whether it was active in that pattern or not, or tell it to only change notes that are currently active in a pattern. The latter is preferable for songwriting but maybe harder to implement from a software point of view.

These presses could be triggered by a external sequencer via MIDI note number (omnni) sent to Carbon and you could have an auto evolving pattern.

This would be better combined with the feature request that can limit Magic to set octaves around a central note ( first note in sequence, as this is normally the root note in a composition).... so +/- octave on magic menu to set it to scale to up and down an octave or more.

* totally stole this form Numerology, but a pattern generator like Carbon is crying out for this feature.

kilpatrickaudio, 2017-07-27:
This is a good idea. I don't mind borrowing ideas from other products... our ideas get borrowed all the time. :)

kilpatrickaudio, 2018-05-01:
We are considering two settings which would be something like MAGIC Range and MAGIC Chance. Range would limit the range of notes (around middle C) and the chance would affect the probability of how many notes would be changed. I'm not sure about how to incorporate the current gate pattern into the decision... this feels like a bit too much "intelligence" for something intended to be random.

hamildad, 2018-05-02:
> Range would limit the range of notes (around middle C)

Would this be tied to track Transpose, so transpose -3 would limit range around A3?

keep Middle C as the baseline but have the range script reference transpose value when adjusting note numbers.

I see this being useful for randomising pitch dependent elements (such as basslines), but not creating too heavy a menu/options list for a simple procedure.

kilpatrickaudio, 2018-05-04:
Fixed in 1.14. Tonality menu has options for Magic Range: semitones) and Magic Change: percent probability of a note being created. Seems cool.

closed 2018-05-04 by kilpatrickaudio